MaJerle/stm32f429 #4: Extern "C" wrapper missing in delay.c and delay.h

JonathanGoldnauVogt, 2016-01-21:
Delay library doesn't work in C++ but doesn't throw linking error due to __weak function definition in STD library. 

Suggested fix is adding wrapper to .c and .h
Thanks.

MaJerle, 2016-01-21:
What is the error?

MaJerle, 2016-01-21:
What is the error?

JonathanGoldnauVogt, 2016-01-21:
There is no error. Which made it harder to debug.
If you download your default project and run it on a discovery-F4 with
parameters "--cpp" the LEDs stay on and do not blink. The systick handler
is not being called properly. I believe the definition of
"__weak void TimingDelay_Decrement" is suppressing a linking error.
Adding extern "C" fixes the issue.

MaJerle, 2016-01-21:
I have added extern "C" in delay library.

Thanks for reporting issue.

closed 2016-01-21 by MaJerle