MaJerle/stm32f429 #6: Definitons?

golddragon007, 2016-04-08:
Hello,

How can I use this in eclipse, because it seems there are some definition which are in the .lib file... Or how can I extract those?

Thanks.

MaJerle, 2016-04-08:
Use what actually?

golddragon007, 2016-04-08:
I want to use the Library 09- I2C and I downloaded from here: http://stm32f4-discovery.com/2014/05/library-09-i2c-for-stm32f4xx/ the TM STM32F4 I2C Library rar file. The problem is, when I add to my project the *.c and *.h files in the eclipse, I get not defined variables, like ADC_SOFTWARE_START . (which is in the tm_stm32_esp8266_keil.lib file, which I can't use, because I using eclipse, not keil...) any solution?

MaJerle, 2016-04-08:
.lib is not for gcc build. Use esp8266at.com for esp library.

closed 2016-04-08 by MaJerle