MaJerle/stm32f429 #10: I want use you lib within stdperiph 1.02

jiuxiaxixi, 2017-06-29:
what will happened if I use a lower stdperiph?

jiuxiaxixi, 2017-06-29:
could I replace some files which I will use?

MaJerle, 2017-06-29:
Im not sure I understand your quesrion.

jiuxiaxixi, 2017-06-29:
Thanks a lot, my problem has been solved!

closed 2017-06-29 by jiuxiaxixi

MaJerle, 2017-06-29:
Why don't you share info what was the problem and what the resolution?

jiuxiaxixi, 2017-06-29:
I am not sure the if the USB Library was compatible with the STM32F4xx_StdPeriph_Driver version 1.02, and I had developed USB2CAN  based on the version 1.02. But I want using your USART with DMA, so I try to upgrade the STM32F4xx_StdPeriph_Driver version to 1.7.

I found that the USB library is independent with the STM32F4xx_StdPeriph_Driver.
Thanks a lot, your work is so fantastic to me!

jiuxiaxixi, 2017-06-29:
And I have something that I thought cool. The SPI driver.
I think the way of using that is so cool
[image]

[rhb_spi_bus_adapter.c.txt](https://github.com/MaJerle/stm32f429/files/1111075/rhb_spi_bus_adapter.c.txt)
[rhb_spi_bus_adapter.h.txt](https://github.com/MaJerle/stm32f429/files/1111078/rhb_spi_bus_adapter.h.txt)

jiuxiaxixi, 2017-06-29:
@MaJerle