MaJerle/stm32f429 #12: Atollic TrueStudio

seyyah, 2018-06-07:
Hi,

How can i use this excellent repo with the Atollic TrueStudio?

MaJerle, 2018-06-07:
Truestusio is general C ARM compiler. This repo are classic c libraries. You can simply add library to project and compile.

SuperFra, 2019-02-05:
Hi Tilen,
congratulations! Very nice work!
I'm new to Stm32 and I would like to learn this fantastic mcu. I'm using the toolchain CubeMx and Atollic TrueStudio. What I'm to set in Cube to generate the right skeleton? 
I know that this is a stupid question for you but I'm new to this world.
Can you help me please?

MaJerle, 2019-02-05:
@SuperFra forget this repository if you are new to STM32 world. By time this libraries were developed, ST already make a step further with development. There is now STM32CubeMX for code generation and (in your case) Atollic TrueSTUDIO for development.

https://www.youtube.com/watch?v=NED_E_z4CwM

SuperFra, 2019-02-05:
Thanks for your quickly answer, but I'm already using it.
I made a webserver with no problem but I would like to serve the pages from sd card. I know write and read from the sd card but I cannot serve the page from there, that's why i was wondering to integrate your library. I need to use some framework like bootstrap and others and "I think" that is not the right choice store the pages in fsdata.c, space apart. Any suggest?
Thanks in advance

MaJerle, 2019-02-05:
So you use LwIP? If so, there is no need to store data in fsdata.c. You can easily implement SD CARD dynamic website using LwIP. Follow LWIP_HTTPD_DYNAMIC_FILE_READ define on Google or LwIP support page. From here, this is not related to STM32 anymore.

SuperFra, 2019-02-05:
Many thanks for your directions! I will try soon.
Bye

McNugget6750, 2020-04-24:
Back to the original question:
Coocox does not exist anymore I have a hard time including your library into Atollic True Studio.
I'm trying to compile the Library 34 to get HID gamepads from my STM32F407 Discovery.

However, when I include the library, I still get undefined references errors:

`
C:\Users\timob\Atollic\TrueSTUDIO\STM32_workspace_9.3\test_3\Debug/..\src/main.c:66: undefined reference to `TM_DISCO_LedInit'
C:\Users\timob\Atollic\TrueSTUDIO\STM32_workspace_9.3\test_3\Debug/..\src/main.c:69: undefined reference to `TM_DISCO_ButtonInit'
C:\Users\timob\Atollic\TrueSTUDIO\STM32_workspace_9.3\test_3\Debug/..\src/main.c:72: undefined reference to `TM_DELAY_Init'
C:\Users\timob\Atollic\TrueSTUDIO\STM32_workspace_9.3\test_3\Debug/..\src/main.c:75: undefined reference to `TM_USB_HIDDEVICE_Init'
C:\Users\timob\Atollic\TrueSTUDIO\STM32_workspace_9.3\test_3\Debug/..\src/main.c:78: undefined reference to `TM_USB_HIDDEVICE_MouseStructInit'
C:\Users\timob\Atollic\TrueSTUDIO\STM32_workspace_9.3\test_3\Debug/..\src/main.c:80: undefined reference to `TM_USB_HIDDEVICE_KeyboardStructInit'
C:\Users\timob\Atollic\TrueSTUDIO\STM32_workspace_9.3\test_3\Debug/..\src/main.c:82: undefined reference to `TM_USB_HIDDEVICE_GamepadStructInit'
C:\Users\timob\Atollic\TrueSTUDIO\STM32_workspace_9.3\test_3\Debug/..\src/main.c:83: undefined reference to `TM_USB_HIDDEVICE_GamepadStructInit'
C:\Users\timob\Atollic\TrueSTUDIO\STM32_workspace_9.3\test_3\Debug/..\src/main.c:87: undefined reference to `TM_USB_HIDDEVICE_GetStatus'
C:\Users\timob\Atollic\TrueSTUDIO\STM32_workspace_9.3\test_3\Debug/..\src/main.c:103: undefined reference to `TM_USB_HIDDEVICE_KeyboardSend'
C:\Users\timob\Atollic\TrueSTUDIO\STM32_workspace_9.3\test_3\Debug/..\src/main.c:113: undefined reference to `TM_USB_HIDDEVICE_KeyboardSend'
`

It seems the source files in the library folder are not being compiled. Only the header files. How can I change this behavior?

[image]
[image]
[image]

Any ideas?

McNugget6750, 2020-04-24:
I brute-forced it to work by compiling and copy and pasting every single file the compiler couldn't find into the src directory until it compiled successfully. Not sure if I got the right files from the large library folder, though, as it doesn't seem to work as intended:

[image]

I only really get a single keyboard and a weird device in my Device Manager. No mouse, gamepad or keyboard shows up within the Device Manager. A keyboard only shows up in the Printers panel but pressing the blue button on the F4 Discovery does nothing.

I guess, I'll continue searching but the example doesn't seem to work just like this. I really want this to work...

closed 2020-12-25 by MaJerle