MaJerle/stm32f429 #18: Is there some place we can contrib to help fix/suggest minor fixes in website/tutorial texts?

utkarshsethi, 2020-04-02:
Now sure what would be the best place to raise this.

https://stm32f4-discovery.net/2014/05/all-stm32f429-libraries-at-one-place/
###Github clone
*If you have any problems porting my libraries into your projects for some reason, you can download my entire repository from `Girhub` here.......*

Is there some way I can suggest/ make fixes while going through the material?

MaJerle, 2020-04-02:
@utkarshsethi what suggestions you have in mind?
Can we use pull requests to solve it? I'm more than happy to migrate changes and new features.

redmasteak1024, 2020-04-03:
Thank you;  Nice working with you .

Tilen Majerle <notifications@github.com>, 2 Nis 2020 Per, 21:28 tarihinde
şunu yazdı:

utkarshsethi, 2020-04-03:
> @utkarshsethi what suggestions you have in mind?
> Can we use pull requests to solve it? I'm more than happy to migrate changes and new features.

I'm going through the tutorials atm, so not much right now.

Pull requests would be fine, but I'm talking about the text you used on the website.  
> Now sure what would be the best place to raise this.
>
> https://stm32f4-discovery.net/2014/05/all-stm32f429-libraries-at-one-place/
> ###Github clone
> _If you have any problems porting my libraries into [...]

e.g. The page linked in my initial comment. has a typo, so this is just a basic minor spelling fix. others might include adding reference links or something maybe? Basically how to contrib to Wikipedia. You fix/add content as you go along

From what I understand you're using WP on the pages so probably not going to work over GitHub :(

P.S.: this should probably be moved to discussions as this is not an 'issue'.

MaJerle, 2020-04-03:
No worries, seems a good discussion.
Yes it is based on WP and the only thing we can do them is to give you an account that has admin priviliges.

For Github docs contribution, something like Sphinx (ReadTheDocs) would be the best, like I have on https://docs.majerle.eu but this would involve HUGE amount of work to transfer all this to a simple files.

utkarshsethi, 2020-04-03:
won't require admin, Wp has various categories, editor, contributor and such,

Another thing you could do is enable people to signup as a predefined limited role, which would allow them to make suggested edits, before going through editor/admin for publishing.

This would also allow a lot more people to contribute.

MaJerle, 2020-04-05:
Quality of the posts must be checked and anyone cannot simply add new one.
Creating an account is feasible, but I need your email and other details.

I propose we continue then over email instead. You can find my email at https://majerle.eu

closed 2020-04-05 by MaJerle