MaJerle/stm32f429 #21: Multiple devices on the same I2C pins

blindemu, 2021-01-10:
Hi, are multiple devices supported on the same I2C pins? I am having trouble reading data from two devices simultaneously, they work just fine individually. Thanks

JIAZM, 2021-01-10:
Have you confirmed that the addresses of connected IIC devices are independent of each other

blindemu, 2021-01-10:
Not sure I understand what you mean, the addresses are 0x52 and 0xD2, is there a reason they would interfere with each other?

JIAZM, 2021-01-10:
The device address of the IIC slave should be 7 bits, with the eighth bit for the read/write command

blindemu, 2021-01-10:
The devices work when the other is disabled (but pins are still connected) which leads me to believe that the addresses are fine? It's only when I try to communicate with both that I see problems arise.

JIAZM, 2021-01-10:
In my experience, you can not cmmunicate with both,  data sending to iic device must be Distinguished with device address