MaJerle/stm32f429 #22: About generate a new size font？

liukai-tech, 2021-01-15:
Hello, I have tested your LCD driver and it works very well, but now I want to generate a larger font(file:tm_stm32f4_fons.c/.h), can you tell me how your font is generated?thank you.

MaJerle, 2021-01-15:
I did not generate fonts by myself, it was part of Google search.
Are you developing strong graphics application? If this is the case, I would advice you to take a look at X-CUBE-TOUCHGFX package from ST. If you are working with SPI-based LCD that has its own memory and have MCU with limited amount of RAM, partial framebuffer is a feature you would go with.

https://support.touchgfx.com/docs/development/scenarios/lowering-memory-usage-with-partial-framebuffer#!

liukai-tech, 2021-01-16:
> I did not generate fonts by myself, it was part of Google search.
> Are you developing strong graphics application? If this is the case, I would advice you to take a look at X-CUBE-TOUCHGFX package [...]

thnak you, I'll check it out.

closed 2021-01-16 by liukai-tech